platformio/platformio-libmirror #3: Support Arduino standard and contributed libraries

ncolomer, 2015-02-10:
Hi @ivankravets, 

Do you think possible to import all the *standard* and *contributed* libraries referenced [here](http://arduino.cc/en/Reference/Libraries) on the official website?

For instance, I'm currently using the `Wire` and I had to manually copy sources into the `lib/` folder of my project. I think it'd be cool to import such "official" libs using platformio's library management.

ivankravets, 2015-02-10:
Hi @ncolomer,

The standard libraries are automatically bundled into Arduino Framework. Do you have any problems?

ncolomer, 2015-02-10:
You're absolutely right... sorry for this dumb issue and thanks for your answer!

What about contributed libraries (such as [MsTimer2](http://playground.arduino.cc/Main/MsTimer2))? If relevant, I can give a hand to import them into platformio.

ivankravets, 2015-02-11:
Sure, I can add them. Can you help me with it? Because I don't have time for it now. As you can see, I'm busy with PlatformIO 1.0.0 release. 

See example of [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html) config https://github.com/ivankravets/platformio-libmirror/blob/master/configs/adafruit/arduino-ht1632.json

I propose you to use "pull-request" to owner of repo. For example, open https://github.com/PaulStoffregen/MsTimer2 then click on [+] near "Branches MsTimer2 / " and type name of file `library.json`. Then, you can grab example of config and edit it for the specific repo. if you will be ready, please inform me, I will look into your fork and correct something if it need. 

What do you think about it? Thanks a lot for helping! :+1:

ncolomer, 2015-02-12:
For sure! Thanks for the detailed procedure, I'll do that this weekend.

ncolomer, 2015-02-15:
(currently working libs json)

Refering the initial subject. I observed that the `EEPROM.h` (seems a [standard library](http://arduino.cc/en/Reference/EEPROM) too) is not automatically bundled into Arduino Framework.

Is it expected?

ivankravets, 2015-02-15:
@ncolomer Do you use the latest `develop` branch?

ivankravets, 2015-02-15:
See here [how to install development version of PlatformIO](http://docs.platformio.org/en/latest/installation.html#development-version).

ncolomer, 2015-02-15:
I'm now running `0.11.0-dev`

In addition to a more verbose output (lot of warning), the `upload` target still fail with the following error:

```
scons: warning: Two different environments were specified for target /Users/ncolomer/Developer/Workspace/perso/arduino/nrf24-server/.pioenvs/arduino_nano/libRadioHead_ID124.a,
	but they appear to have the same action: $AR $ARFLAGS $TARGET $SOURCES
$RANLIB $RANLIBFLAGS $TARGET
File "/Library/Python/2.7/site-packages/platformio/builder/tools/platformio.py", line 107, in BuildLibrary
.pioenvs/arduino_nano/RadioHead_ID124/STM32ArduinoCompat/HardwareSPI.cpp:6:20: fatal error: wirish.h: No such file or directory
 #include <wirish.h>
                    ^
compilation terminated.
scons: *** [.pioenvs/arduino_nano/RadioHead_ID124/STM32ArduinoCompat/HardwareSPI.o] Error 1
```

Surprisingly, the `build` target seems to only output warnings (ends with a success).
I took care to run a `clean`before those tests.

ivankravets, 2015-02-15:
Could you report this issue to PlatformIO repo with `src` details? I will test it tomorrow. Thanks :+1:

ncolomer, 2015-02-16:
Will do this noon
Thanks a lot!

ivankravets, 2015-03-12:
If you still have any problems with the latest 1.1.0 release, please open new issue in PlatformIO repo. Thanks :+1:

closed 2015-03-12 by ivankravets