platformio/platformio-libmirror #7: Library #127 officially discarded by author

gandy92, 2015-03-11:
The README.md of library #127 (ESP8266) states:

This library has been discarded. A new library named WeeESP8266 is recommended, which is more easy-to-use for users.
WeeESP8266 can be downloaded at <https://github.com/itead/ITEADLIB_Arduino_WeeESP8266>. 

Do you think it's possible to also add the new library?

ivankravets, 2015-03-11:
Done, see https://github.com/itead/ITEADLIB_Arduino_WeeESP8266/pull/8

P.S: The http://platformio.org/#!/lib/show/127/ESP8266 has been updated with the new version.

gandy92, 2015-03-11:
Cool, thanks! Amazing :-)

ivankravets, 2015-03-11:
If you are interested in adding new libraries, I would be thankful. I can explain how do it. I have a huge list with the future libraries, but don't have time for them :cry: 

Thanks :blush:

gandy92, 2015-03-11:
There actually is a list of libraries I would like to see supported by PlatformIO and I'd like to help where I can. I do need some assistance though. A few questions that come to mind:
- Several of the libraries I use are sitting in one single git repository. Would they be added with one json file or do I need to provide multiple files?
- Do I need to clone a repository in order to add a library.json file or can I just write put them somewhere?
- What about licences?
- Is there any way to add a new board to the arduino framework? (e.g. the http://www.panstamp.com/)

I'm sure there will be more.. how do I start? :-)

ivankravets, 2015-03-11:
> Is there any way to add a new board to the arduino framework? (e.g. the http://www.panstamp.com/)

Sure, just open separate issue in PlatformIO repo https://github.com/ivankravets/platformio/issues

Please include that information: 
* http://www.panstamp.com/
* https://github.com/panStamp/panstamp/wiki/panStamp%20AVR.-Technical%20details

ivankravets, 2015-03-12:
> There actually is a list of libraries I would like to see supported by PlatformIO and I'd like to help where I can. I do need some assistance though.

1. [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html) allows to describe THE ONLY ONE library. This is manifest file of the specific library. External library managers use these files. However,  you can have multiple libraries in the one repository. In this case, you should create separated ``library.json`` files in the specific folder of each library. See https://github.com/jrowberg/i2cdevlib project, which contains multiple libs and multiple ``library.json`` and result:
http://platformio.org/#!/lib/search?query=keyword%253A%2522i2cdevlib%2522

For example, https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/AD7746/library.json
Here, ``"include": "Arduino/AD7746",`` option describes where is located source code of library.

2. You can pull your ``library.json`` files direct to libraries owners. See my pull-requests history: https://github.com/ivankravets?tab=contributions&period=monthly
Just follow to your favourite library repository, click "+" in the root of library and add new file named ``library.json``. Then, fill it (many examples are in my pull-requests or use documentation or ``i2cdev`` libraries), then click "Create pull-requst". When you will create it, please write here, I will look on it and explain what you should do for next.

3. You will not modify any code, you will just propose ``library.json``. The you will use "RAW" code of this file and register via ``platformio lib register http://url/to/library.json`` command. See documentation http://docs.platformio.org/en/latest/librarymanager/creating.html

gandy92, 2015-03-12:
Thanks for the instructions, I think I can go from there :) I'll be away from civilization (read: internet access) next week, so don't expect anything too soon.

ivankravets, 2015-03-12:
No problems :blush: Have a nice vocations!

gandy92, 2015-03-12:
Thanks - anyway, couldn't wait, so I started with cloning https://github.com/watterott/Arduino-Libs to https://github.com/gandy92/Arduino-Libs
- I wrote a shell script that creates a number of library.json files, all with description (taken from the README.md) and keywords (from an auxiliary file).
- I read through the files and could not find any obvious errors, but please, check yourself. 
- I omitted any sub-library that obviously originates from another github project

Files I created:
- ADS1147/library.json
- ADS7846/library.json
- DAC8760/library.json
- digitalWriteFast/library.json
- DisplayI2C/library.json
- DisplaySPI/library.json
- DS1307/library.json
- GraphicsLib/library.json
- MI0283QT2/library.json
- MI0283QT9/library.json
- mSD_Shield/library.json
- RedFly/library.json
- RotaryEncoder/library.json
- RV8523/library.json
- S65L2F50/library.json
- S65LPH88/library.json
- S65LS020/library.json
- SSD1331/library.json

Can you check if I made any mistakes, before I create a pull-request?

Thanks

ivankravets, 2015-03-12:
Wow! :open_mouth:  Excellent job! :+1: 

A few corrections/propositions:
1. Please remove prefix ``Watterott-`` from library name.  We don't use owners of lib within library names.
2. ``"include": "Arduino/ADS7846",`` - why do you use ``Arduino/`` folder? Where do you see it? It should be ``"include": "ADS7846",``. "include" means "relative path" to source code of library within repository.

Can you fix that and report here? Then I will re-check again! Thanks!

gandy92, 2015-03-12:
Thank you for checking and more so for your feedback
1. Ok
2. My bad - copy and paste without really thinking..
All changes applied, please re-check, thanks!

ivankravets, 2015-03-12:
1. Adafruit likes to name own libraries with "Adafruit-" prefix. The even name "include" files like `Adafruit_BlaBla.h`. If you want to add "Watterott-" like a prefix - NO PROBLEM. We are open source, and each of us decide how to name own libraries.
2. Oopss... :smile: You have mismatches between lib names. Original names look without "Watterrott", but `dependencies` want with that prefix. 

Again, please decide how do you want to name these libs. In my vision, is good to give _TECHNICAL_ name which will be informative for end user (without "Adafruit, Watterott..."). Each libs have "author" information and PlatformIO shows this info. 

The `library.json` has not any relations to PlatformIO Library Manager. PlatformIO proposes it like a  standard to describe embedded libraries. The owner of `library.json` is owner of the lib. He is responsible for the all informations, including "name", "description", "keywords" and etc.

gandy92, 2015-03-13:
Thank you again :-)
1. Ok, I do see the point now, and I agree. Besides, we do need some common ground.
2. Another silly mistake, should be fixed now. Sorry for the inconvenience.

Please, recheck one more time...

ivankravets, 2015-03-13:
2. I've not seen any changes yet. 

For example, 
https://github.com/gandy92/Arduino-Libs/blob/master/ADS1147/library.json

```json
  "dependencies":
  {
    "name": "Watterott-digitalWriteFast",
    "frameworks": "arduino"
  },
```

I mean, there is no here ``Watterott-digitalWriteFast`` library. See https://github.com/gandy92/Arduino-Libs/blob/master/digitalWriteFast/library.json

ivankravets, 2015-03-13:
Here is really STRICT dependency:

It should look like:
```json
  "dependencies":
  {
    "name": "digitalWriteFast",
    "frameworks": "arduino",
    "authors": "Watterott electronic"
  },
```

gandy92, 2015-03-13:
Sorry, I was in a hurry this morning and forgot to push after the commit.. did this now and checked with the github website, the changes are in now.

ivankravets, 2015-03-13:
As for me, it looks good now. I propose you to "pull request" to original repo, then I will explain how to register each lib. Tell me please when pull request will be ready.

gandy92, 2015-03-13:
Perfect. I created the pull request (https://github.com/watterott/Arduino-Libs/pull/4), please advise :-)

ivankravets, 2015-03-13:
I see that @watterott has just merged your request. Thanks!

Now, you can register these libs via
```shell
> platformio lib register %LIB_JSON_URL%
```

For example, please follow to this config:
https://github.com/watterott/Arduino-Libs/blob/master/ADS1147/library.json

Then click on "RAW" button, and copy-paste this URL. The result:
```shell
> platformio lib register https://raw.githubusercontent.com/watterott/Arduino-Libs/master/ADS1147/library.json
> #The library has been successfully registered and is waiting for moderation
```

I've just registered ADS1147 library. Please register the rest libs.

Again, thanks a lot  for great contribution! :+1:

gandy92, 2015-03-13:
Done :-) Thanks again for your help and your patience! 

Guess I'm ready now to help you working on your list of libraries you mentioned earlier.

ivankravets, 2015-03-13:
Works great!
```shell
> platformio lib search ADS1147
Found 1 libraries:

[ ID  ] Name             Compatibility         "Authors": Description
-------------------------------------------------------------------------------------
[ 139 ] ADS1147          arduino, atmelavr     "Watterott electronic": ADS1147 ADC (Analog-Digital-Converter)

> platformio lib install 139
Installing library [ 139 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #139 'ADS1147' has been successfully installed!
Installing dependencies:
Installing library [ 142 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #142 'digitalWriteFast' has been successfully installed!
```

P.S: We should be thankful @watterott too :blush:

ivankravets, 2015-03-13:
> Guess I'm ready now to help you working on your list of libraries you mentioned earlier.

I've just opened new issue. Let's continue out discussion here https://github.com/ivankravets/platformio-libmirror/issues/8

closed 2015-03-13 by ivankravets