platformio/platformio-libmirror #14: TMRh20's RF24 library

ncolomer, 2015-06-16:
Hi guys,
I'd like to add RF24 library to the PlatformIO registry. It is hosted on Github https://github.com/TMRh20/RF24 but the structure is not common: there is a `arch` directory containing implementations by platform.
I'm wondering if a standard `library.json` is enough or if we have to add additional metadata to the `json` file to hints about differnt implems/archs?
Thanks!

ivankravets, 2015-06-18:
You need to use ``include`` field from [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html#include). See my pull-requst https://github.com/TMRh20/RF24/pull/122 for @TMRh20

You can try different examples for the multiple boards (Uno, Teensy, Due) using [PlatformIO Continuous Integration](http://docs.platformio.org/en/latest/ci/index.html) shortcut:
```shell
platformio lib install 433  # http://platformio.org/#!/lib/show/433/RF24
platformio ci ~/.platformio/lib/RF24_ID433/examples/GettingStarted --board uno --board teensy31 --board due
```

ncolomer, 2015-06-22:
Great! Thanks for the PR and explanations ;)

closed 2015-06-22 by ncolomer