platformio/platformio-libmirror #16: What is the process for moving libraries from moderated directory?

jrobeson, 2015-09-06:
The one lib I need is in the moderated directory (BLE_API). Is there any instructions for how one can help move a library along?

ivankravets, 2015-09-06:
Where is located the source code of this library?

jrobeson, 2015-09-06:
Here's the platformio defintion: https://github.com/platformio/platformio-libmirror/blob/master/configs/mbed/moderation/BLE_API_BluetoothLowEnergy.json 

and here's the library source code: https://developer.mbed.org/teams/Bluetooth-Low-Energy/code/BLE_API/

ivankravets, 2015-09-06:
1. Are you going to modify source code of this lib?
2. If yes, do you want share your modified lib with others?

If no, you can clone it to `lib\BLE_API` and make modifications.

jrobeson, 2015-09-06:
I just want to be able to install it with platformio lib install and have it registered as a library

jrobeson, 2015-09-06:
Can you tell me why it is in that moderated directory and how to get it out of there?

jrobeson, 2015-09-06:
i should also mention that the library is in your top MBED TODO list as seen here: https://github.com/platformio/platformio-libmirror/blob/master/scripts/mbed-top/TOP-libs-TODO.md

jrobeson, 2015-09-06:
I can say that simply cloning BLE_API isn't enough to get it linking out of the box (like i assume other libraries do). My program compiles fine with the MBED online compiler, but not with `platformio run`. It doesn't look like there are enough include paths compared to what i see in the online compiler output.

I see 
```
Sun Sep  6 05:08:45 2015] Processing nrf51_dk (platform: nordicnrf51, targets: upload, framework: mbed, board: nrf51_dk)
--------------------------------------------------------------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/nrf51_dk/src/main.o -c -std=gnu++98 -fno-rtti -fdata-sections -ffunction-sections -Wno-unused-parameter -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -mcpu=cortex-m0 -D__CORTEX_M0 -DTARGET_NRF51_DK -DTARGET_FF_ARDUINO -DTOOLCHAIN_GCC_ARM -DTOOLCHAIN_GCC -DTARGET_MCU_NRF51822 -DTARGET_MCU_NORDIC_32K -DMBED_BUILD_TIMESTAMP=1441274200.42 -DNRF51 -DTARGET_CORTEX_M -DTARGET_NRF51822 -DARM_MATH_CM0 -DTARGET_NORDIC -DTARGET_MCU_NRF51_32K -DTARGET_MCU_NRF51 -DTARGET_M0 -D__MBED__=1 -DPLATFORMIO=020300 -I.pioenvs/nrf51_dk/FrameworkMbedInc248832578 -I.pioenvs/nrf51_dk/FrameworkMbedInc-213564679 -I.pioenvs/nrf51_dk/FrameworkMbedInc-250686161 -I.pioenvs/nrf51_dk/FrameworkMbedInc1043689724 -I.pioenvs/nrf51_dk/FrameworkMbedInc-595756208 -I.pioenvs/nrf51_dk/FrameworkMbedInc467943721 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1537068880 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1425851379 -I.pioenvs/nrf51_dk/FrameworkMbedInc992684671 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1635307249 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1354727534 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1881932505 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1156658389 -I.pioenvs/nrf51_dk/FrameworkMbedInc2046170798 -I.pioenvs/nrf51_dk/FrameworkMbedInc-476856032 -I.pioenvs/nrf51_dk/FrameworkMbedInc985609110 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1188180368 -I.pioenvs/nrf51_dk/FrameworkMbedInc633976207 -I.pioenvs/nrf51_dk/FrameworkMbedInc1864696863 -I.pioenvs/nrf51_dk/FrameworkMbedInc1692272459 -I.pioenvs/nrf51_dk/FrameworkMbedInc39259746 -I.pioenvs/nrf51_dk/FrameworkMbedInc800437392 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1178622005 -I.pioenvs/nrf51_dk/BLE_API src/main.cpp
arm-none-eabi-ar rcs .pioenvs/nrf51_dk/libFrameworkMbed.a /home/johnny/.local/share/platformio/packages/framework-mbed/variant/NRF51_DK/mbed/TARGET_NRF51_DK/TOOLCHAIN_GCC_ARM/cmsis_nvic.o /home/johnny/.local/share/platformio/packages/framework-mbed/variant/NRF51_DK/mbed/TARGET_NRF51_DK/TOOLCHAIN_GCC_ARM/retarget.o /home/johnny/.local/share/platformio/packages/framework-mbed/variant/NRF51_DK/mbed/TARGET_NRF51_DK/TOOLCHAIN_GCC_ARM/system_nrf51.o /home/johnny/.local/share/platformio/packages/framework-mbed/variant/NRF51_DK/mbed/TARGET_NRF51_DK/TOOLCHAIN_GCC_ARM/system_nrf51822.o /home/johnny/.local/share/platformio/packages/framework-mbed/variant/NRF51_DK/mbed/TARGET_NRF51_DK/TOOLCHAIN_GCC_ARM/board.o /home/johnny/.local/share/platformio/packages/framework-mbed/variant/NRF51_DK/mbed/TARGET_NRF51_DK/TOOLCHAIN_GCC_ARM/startup_NRF51822.o
arm-none-eabi-ranlib .pioenvs/nrf51_dk/libFrameworkMbed.a
arm-none-eabi-g++ -o .pioenvs/nrf51_dk/BLE_API/source/BLE.o -c -std=gnu++98 -fno-rtti -fdata-sections -ffunction-sections -Wno-unused-parameter -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -mcpu=cortex-m0 -D__CORTEX_M0 -DTARGET_NRF51_DK -DTARGET_FF_ARDUINO -DTOOLCHAIN_GCC_ARM -DTOOLCHAIN_GCC -DTARGET_MCU_NRF51822 -DTARGET_MCU_NORDIC_32K -DMBED_BUILD_TIMESTAMP=1441274200.42 -DNRF51 -DTARGET_CORTEX_M -DTARGET_NRF51822 -DARM_MATH_CM0 -DTARGET_NORDIC -DTARGET_MCU_NRF51_32K -DTARGET_MCU_NRF51 -DTARGET_M0 -D__MBED__=1 -I.pioenvs/nrf51_dk/FrameworkMbedInc248832578 -I.pioenvs/nrf51_dk/FrameworkMbedInc-213564679 -I.pioenvs/nrf51_dk/FrameworkMbedInc-250686161 -I.pioenvs/nrf51_dk/FrameworkMbedInc1043689724 -I.pioenvs/nrf51_dk/FrameworkMbedInc-595756208 -I.pioenvs/nrf51_dk/FrameworkMbedInc467943721 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1537068880 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1425851379 -I.pioenvs/nrf51_dk/FrameworkMbedInc992684671 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1635307249 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1354727534 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1881932505 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1156658389 -I.pioenvs/nrf51_dk/FrameworkMbedInc2046170798 -I.pioenvs/nrf51_dk/FrameworkMbedInc-476856032 -I.pioenvs/nrf51_dk/FrameworkMbedInc985609110 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1188180368 -I.pioenvs/nrf51_dk/FrameworkMbedInc633976207 -I.pioenvs/nrf51_dk/FrameworkMbedInc1864696863 -I.pioenvs/nrf51_dk/FrameworkMbedInc1692272459 -I.pioenvs/nrf51_dk/FrameworkMbedInc39259746 -I.pioenvs/nrf51_dk/FrameworkMbedInc800437392 -I.pioenvs/nrf51_dk/FrameworkMbedInc-1178622005 -I.pioenvs/nrf51_dk/BLE_API .pioenvs/nrf51_dk/BLE_API/source/BLE.cpp
arm-none-eabi-g++ -o .pioenvs/nrf51_dk/BLE_API/source/DiscoveredCharacteristic.o -c -std=gnu++98 -fno-rtti -fdata-sections -ffunction-sections -Wno-unused-parameter -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -mcpu=cortex-m0 -D__CORTEX_M0 -DTARGET_NRF51_DK -DTARGET_FF_ARDUINO -DTOOLCHAIN_GCC_ARM -DTOOLCHAIN_GCC -DTARGET_MCU_NRF518 [... cut ...] 51_dk/libFrameworkMbed.a .pioenvs/nrf51_dk/libBLE_API.a -Wl,--end-group
.pioenvs/nrf51_dk/src/main.o: In function `_GLOBAL__sub_I_ble':
main.cpp:(.text.startup._GLOBAL__sub_I_ble+0x2): undefined reference to `createBLEInstance()'
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/nrf51_dk/firmware.elf] Error 1
```

There is a header for `createBLEInstance` in BLE_API, but the actual definition is from this library: https://developer.mbed.org/teams/Nordic-Semiconductor/code/nRF51822/

jrobeson, 2015-09-06:
looks like I'll have to add some includes to the file.

jrobeson, 2015-09-06:
my question was answered in this issue https://github.com/platformio/platformio-libmirror/issues/10

closed 2015-09-06 by jrobeson