platformio/platformio-libmirror #18: Please add following libraries

squix78, 2016-01-17:
https://github.com/squix78/json-streaming-parser
https://github.com/squix78/esp8266-weather-station
https://github.com/squix78/esp8266-oled-ssd1306

I have added them using the register keyword from the command line but I'm not sure if that is the proper way to submit/request a library to be added...

thank you, Dani

ivankravets, 2016-01-18:
1.Thanks a lot for your library.json. I've tried to disable moderation and appied your library.json without review as people requested https://github.com/platformio/platformio/issues/440
2. Please fix all names in your libs according the rest 300 libs. I can't image how user will `platformio lib install "ESP8266 Weather Station"` in PlatformIO 3.0. It should be `platformio install WeatherStation`.
3. Please remove problematic keywords with trademarks. Like, "arduino", "esp8266" and etc. Users use keywords to filter/categorizes libraries.

ivankravets, 2016-01-18:
See my PR
* https://github.com/squix78/esp8266-oled-ssd1306/pull/18
* https://github.com/squix78/esp8266-weather-station/pull/16
* https://github.com/squix78/json-streaming-parser/pull/1

I see that need to improve [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html) specification and explain the rules from `name` field. The library.json submitter should know that later other developers can depend on this lib by this "long name".

ivankravets, 2016-02-28:
@squix78 please merge my PRs. Thanks! https://github.com/squix78/esp8266-oled-ssd1306/pull/18

closed 2016-02-28 by ivankravets