platformio/platformio-libmirror #21: [enhancement] Add Adafruit MMA8451 library

NetForces, 2016-02-18:
Used for the MMA8451 Accelerator breakout board

https://github.com/adafruit/Adafruit_MMA8451_Library

NetForces, 2016-02-18:
Wanted to make a branch and submit a PR but seems I can not... So here is arduino-MMA8451.json

```
{
  "name": "Adafruit-MMA8451",
  "keywords": "accelerometer",
  "description": "Arduino library for the MMA8451 Triple-Axis Accelerometer - ±2/4/8g @ 14-bit",
  "repository":
  {
    "type": "git",
    "url": "https://github.com/adafruit/Adafruit_MMA8451_Library"
  },
  "frameworks": "arduino",
  "platforms": "atmelavr"
}
```

ivankravets, 2016-02-18:
Added https://github.com/platformio/platformio-libmirror/blob/master/configs/adafruit/MMA8451.json

You can make PR to this repo https://github.com/platformio/platformio-libmirror/tree/master/configs

closed 2016-02-18 by ivankravets

ivankravets, 2016-02-18:
Registry: http://platformio.org/#!/lib/show/1099/Adafruit-MMA8451