platformio/platformio-libmirror #22: Compile error trying to use Adafruit_MMA8451

NetForces, 2016-02-18:
I installed it just fine:

```
Andres-MacBook-Pro:Arduino courchea$ platformio lib list
[ ID  ] Name             Compatibility         "Authors": Description
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[ 28  ] Adafruit-NeoPixel arduino, atmelavr     "Adafruit Industries": Arduino library for controlling single-wire-based LED pixels and strip
[ 31  ] Adafruit-Unified-Sensor arduino, atmelavr     "Adafruit Industries": Adafruit Unified Sensor Driver
[1099 ] Adafruit-MMA8451 arduino, atmelavr     "Adafruit Industries": Arduino library for the MMA8451 Triple-Axis Accelerometer - 2/4/8g @ 14-bit
```

However getting a error on compile:

```
Andres-MacBook-Pro:Arduino courchea$ platformio run
[Thu Feb 18 07:54:21 2016] Processing uno (platform: atmelavr, board: uno, framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/uno/src/tmp_ino_to.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -DPLATFORMIO=020804 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/Wire -I.pioenvs/uno/Wire/utility -I.pioenvs/uno/Adafruit_Unified_Sensor_ID31 src/tmp_ino_to.cpp
src/tmp_ino_to.cpp:4:30: fatal error: Adafruit_MMA8451.h: No such file or directory
#include <Adafruit_MMA8451.h>
^
compilation terminated.
scons: *** [.pioenvs/uno/src/tmp_ino_to.o] Error 1
==================================================================================================== [ ERROR ] Took 0.54 seconds ====================================================================================================
```

valeros, 2016-02-18:
Hi @NetForces!
I've just compiled library examples without any error. Could you share your project as .zip somewhere?

NetForces, 2016-02-18:
Let me make a spin-off minimalist of it as it contains a lot of things.

NetForces, 2016-02-18:
Forget that, works. Had local copies of Adafruit_MMA8451.h and Adafruit_MMA8451.cpp which were conflicting.

closed 2016-02-18 by NetForces