platformio/platformio-libmirror #44: Add PJON library

ivankravets, 2016-03-28:
https://github.com/gioblu/PJON

valeros, 2016-03-28:
Done http://platformio.org/lib/show/245/PJON

gepd, 2016-03-28:
@valeros I tried to compile the first example in the project's github but it output some errors:

```
Processing megaatmega2560 (platform: atmelavr, board: megaatmega2560, framework: arduino)

avr-g++ -o .pioenvs\megaatmega2560\src\tmp_ino_to.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10608 -DPLATFORMIO=020806 -I.pioenvs\megaatmega2560\FrameworkArduino -I.pioenvs\megaatmega2560\FrameworkArduinoVariant -I.pioenvs\megaatmega2560\PJON_ID245 F:\Dropbox\Examples\pde-test\tmp_ino_to.cpp
z-main.pde: In function 'void setup()':
z-main.pde:5:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
In file included from a-test.pde:3:0:
.pioenvs\megaatmega2560\PJON_ID245/PJON.h: At global scope:
.pioenvs\megaatmega2560\PJON_ID245/PJON.h:142:8: error: redefinition of 'struct packet'
struct packet {
^
In file included from F:\Dropbox\Examples\pde-test\tmp_ino_to.cpp:2:0:
.pioenvs\megaatmega2560\PJON_ID245/PJON.h:142:8: error: previous definition of 'struct packet'
struct packet {
^
In file included from a-test.pde:3:0:
.pioenvs\megaatmega2560\PJON_ID245/PJON.h: In function 'void dummy_error_handler(uint8_t, uint8_t)':
.pioenvs\megaatmega2560\PJON_ID245/PJON.h:155:13: error: redefinition of 'void dummy_error_handler(uint8_t, uint8_t)'
static void dummy_error_handler(uint8_t code, uint8_t data) {};
^
In file included from F:\Dropbox\Examples\pde-test\tmp_ino_to.cpp:2:0:
.pioenvs\megaatmega2560\PJON_ID245/PJON.h:155:13: error: 'void dummy_error_handler(uint8_t, uint8_t)' previously defined here
static void dummy_error_handler(uint8_t code, uint8_t data) {};
^
In file included from a-test.pde:3:0:
.pioenvs\megaatmega2560\PJON_ID245/PJON.h: In function 'void dummy_receiver_handler(uint8_t, uint8_t*)':
.pioenvs\megaatmega2560\PJON_ID245/PJON.h:156:13: error: redefinition of 'void dummy_receiver_handler(uint8_t, uint8_t*)'
static void dummy_receiver_handler(uint8_t length, uint8_t *payload) {};
^
In file included from F:\Dropbox\Examples\pde-test\tmp_ino_to.cpp:2:0:
.pioenvs\megaatmega2560\PJON_ID245/PJON.h:156:13: error: 'void dummy_receiver_handler(uint8_t, uint8_t*)' previously defined here
static void dummy_receiver_handler(uint8_t length, uint8_t *payload) {};
^
In file included from a-test.pde:3:0:
.pioenvs\megaatmega2560\PJON_ID245/PJON.h: At global scope:
.pioenvs\megaatmega2560\PJON_ID245/PJON.h:158:7: error: redefinition of 'class PJON'
class PJON {
^
In file included from F:\Dropbox\Examples\pde-test\tmp_ino_to.cpp:2:0:
.pioenvs\megaatmega2560\PJON_ID245/PJON.h:158:7: error: previous definition of 'class PJON'
class PJON {
^
.pioenvs\megaatmega2560\PJON_ID245/PJON.h:155:13: warning: 'void dummy_error_handler(uint8_t, uint8_t)' defined but not used [-Wunused-function]
static void dummy_error_handler(uint8_t code, uint8_t data) {};
^
.pioenvs\megaatmega2560\PJON_ID245/PJON.h:156:13: warning: 'void dummy_receiver_handler(uint8_t, uint8_t*)' defined but not used [-Wunused-function]
static void dummy_receiver_handler(uint8_t length, uint8_t *payload) {};
^
scons: *** [.pioenvs\megaatmega2560\src\tmp_ino_to.o] Error 1
```

valeros, 2016-03-29:
Hi @gepd !
I compiled some examples `BlinkTest`, `ErrorTest` from the library and all of them work fine.
`error: previous definition of` error points that you have the same code in different places.
Could you please create separate project with only this [file](https://github.com/gioblu/PJON/blob/master/examples/BlinkTest/Receiver/Receiver.ino) in the `src` directory and install the library globally via `platformio lib install 245`

gepd, 2016-03-29:
Thanks @valeros as you said the example works without problem, the error is because the project has multiples files. So you can can close this issue.

closed 2016-03-31 by ivankravets