platformio/platformio-libmirror #45: Add Libraries rc-switch, VirtualWire

TheOneValen, 2016-03-29:
Hi!

I am proposing the following libraries for the repository:

https://www.pjrc.com/teensy/td_libs_VirtualWire.html
https://github.com/sui77/rc-switch

Greetings.

valeros, 2016-03-30:
Done!
http://platformio.org/lib/show/251/VirtualWire
http://platformio.org/lib/show/246/RCSwitch

closed 2016-03-30 by valeros

TheOneValen, 2016-03-31:
Wow that was fast. To imagine what time I could have saved with platformio compared to fiddling with Makefiles...

There is a newer version available for VirtualWire (Final one)
http://www.airspayce.com/mikem/arduino/VirtualWire/VirtualWire-1.27.zip

Which is the only lib I would need to fully migrate to platformio and ditch my makefiles...