platformio/platformio-libmirror #62: NTPClient has moved?

SirUli, 2016-05-14:
Hello,

It looks as if http://platformio.org/lib/show/551/NTPClient has moved to https://github.com/arduino-libraries/NTPClient

It became evident when downloading the library 551 which does not contain the same functions and behaves not according to the examples (e.g. getEpochTime() is missing).

As the Link https://github.com/FWeinb/NTPClient.git points to https://github.com/arduino-libraries/NTPClient i'd assume that it has been moved there?

Best Regards,
Uli

ivankravets, 2016-05-14:
Thanks! Let's wait when @FWeinb merge PR https://github.com/arduino-libraries/NTPClient/pull/18 .

ivankravets, 2016-05-25:
Updated http://platformio.org/lib/show/551/NTPClient

closed 2016-05-25 by ivankravets