platformio/platformio-libmirror #67: ADS1x15 lib

aShure, 2016-06-23:
gentle request: https://github.com/soligen2010/Adafruit_ADS1X15

The original version at Adafruit isn't maintained well, this one includes a couple pull requests. Adafruit have got a long list of open pull requests for some of their libs 👎

valeros, 2016-06-24:
Added http://platformio.org/lib/show/344/ADS1X15
Thanks!

closed 2016-06-24 by valeros

aShure, 2016-06-24:
I saw you added 342 with the original Adafruit lib and this more advanced 344, but somehow the content seems to be the same. The 344 doesn't contain the functions from the linked lib but only the standard Adafruit ones.

ivankravets, 2016-06-25:
@aShure the manifest file links to the release 1.0.0. Check it https://github.com/soligen2010/Adafruit_ADS1X15/releases

There 2 options in this situation:
1. To ask @soligen2010 to make new release and update `library.json` version field.
2. Directly update `version` field and don't make release (don't recommend)
3. Remove `version` field from manifest and in this case PlatformIO Crawler will use the latest commit as a "new version". Also, don't recommend to use it because we will ask all developers to make versioned libraries. PlatformIO 3.0 is coming...

soligen2010, 2016-06-28:
The library Json can be updated. You can make a pull request to my fork or I can do it. I don't know what the platform IO processes is, so I  don't know all that needs to be done. Is there more to defining a release that updating the json?

ivankravets, 2016-06-29:
@soligen2010 are you going to make new releases? If not, need to remove `version` field from manifest. In this case, PlatformIO will update the source code of the library on the each commit.

soligen2010, 2016-06-29:
I cant figure out how to define a release.  Maybe because I am a fork and not the original? So, I removed the version from the JSON so hopefully that fixes eShure's issue

soligen2010, 2016-06-29:
I found where to define a release, so I made a release 1.2.0 and put the release back into the JSON.

soligen2010, 2016-06-29:
I updated the JSON in my project, but I noticed that there is one in this project as well.  Does it need to be updated too?

ivankravets, 2016-06-29:
Library will be updated automatically within 24h. Thanks.

ivankravets, 2016-06-30:
I that library has been updated. http://platformio.org/lib/show/344/ADS1X15

@soligen2010 could you confirm that?

soligen2010, 2016-06-30:
Yes, it now has my modified code.  Thanks.