platformio/platformio-libmirror #87: Add support for Queue & Stack List & Array libraries

mohsh86, 2016-08-31:
there are libraries that are in Arduino's playground code site,

Library implementing a generic, dynamic stack ([array version](http://playground.arduino.cc/Code/StackArray))
Library implementing a generic, dynamic stack ([linked list version](http://playground.arduino.cc/Code/StackList))
Library implementing a generic, dynamic queue ([array version](http://playground.arduino.cc/Code/QueueArray))
Library implementing a generic, dynamic queue ([linked list version](http://playground.arduino.cc/Code/QueueList))

these are zip and do not have corrosponding github, would you bother of keeping such libs in the registry ? or shall someone install it manually ? and is there a feature to install lib from zip in platformio lib console ?

cheers

mohsh86, 2016-09-01:
its worth mentioning that i've tried install them with pio lib -g install [zip url] and it couldn't unpack since there is no .json file in those archives.

ivankravets, 2016-09-01:
Technically, PlatformIO 3.0 Library Manager supports packages with manifest or plain VCS repositories. Otherwise, you can place this library to storage by self.

Nevertheless, please open issue here https://github.com/platformio/platformio/issues. I'll try to implement installing from RAW archives without manifests.

closed 2016-09-02 by mohsh86