platformio/platformio-libmirror #90: Adafruit libraries deps broken

zgoda, 2016-09-01:
```
$ pio --version
PlatformIO, version 3.0.0b7
$ pio lib -g install 81
Library Storage: /home/jazg/.platformio/lib
LibraryManager: Installing id=81
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Adafruit-PCD8544 @ 5c385af945 has been successfully installed!
Installing dependencies
Looking for Adafruit-GFX library in registry
Error: Library `{u'frameworks': [u'arduino'], u'name': u'Adafruit-GFX'}` has not been found in the registry
```

zgoda, 2016-09-01:
Adafruit-GFX is most probably now called "Adafrit GFX Library" -> http://platformio.org/lib/show/13/Adafruit%20GFX%20Library

zgoda, 2016-09-01:
All of sudden, projects that use Arduino framework now fail to build or even clean:

```
$ pio run -t clean
[Thu Sep  1 16:12:37 2016] Processing nodemcuv2 (platform: espressif, board: nodemcuv2, framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 38 compatible libraries
Looking for dependencies...
Error: Could not find `Adafruit-GFX` dependency for `Adafruit-PCD8544` library
============================================== [ERROR] Took 0.47 seconds ==============================================
```

They start working again if I remove library id 81 that has broken dependencies.

ivankravets, 2016-09-01:
Fixed https://github.com/WereCatf/Adafruit-PCD8544-Nokia-5110-LCD-library/pull/2

zgoda, 2016-09-01:
This is not the same library. The one I'm talking about is Adafruit original, id 81. Your PR is against library id 257 (optimized for ESP8266 and STM32).

ivankravets, 2016-09-01:
Thanks, fixed!

closed 2016-09-01 by ivankravets