platformio/platformio-libmirror #91: Missing "'SPI.h"

mohsh86, 2016-09-04:
SPI.h is avr library usually located in <arduino-1.6.xx>/hardware/arduino/avr/libraries/SPI/src

problem was in PlatformIO 2.11, when i frst installed 3.0.0 ( i think 2b maybe), the problem was gone, now that i upgraded to 3.0.0.13, the problem is back again.

ivankravets, 2016-09-04:
We disabled deep dependency finder in latest builds of PlatformIO 3.0 to avoid issues with old Arduino libraries. 

Please read http://docs.platformio.org/en/latest/librarymanager/ldf.html

mohsh86, 2016-09-04:
anyway to avoid the issue ? SPI.h is required by RFM69 Library (lib id 92), it isn't something in my code..

ivankravets, 2016-09-04:
You have a few options:

1. To turn DEEP search for LIbrary Dependency Finder (`lib_ldf_mode = 2`). See docs above
2. To add `#include <SPI.h>` to the one file of your project.
3. Developer of RFM69 should include `SPI.h` in main `RFM69.h`. In this case PlatformIO will find nested includes in LDF MODE = 1

mohsh86, 2016-09-04:
i've included lib_force=SPI in project config under [env], is that a correct way as well ?

mohsh86, 2016-09-04:
i've changed lib_ldf_mode=2 and now it works as well, thanks :+1:

ivankravets, 2016-09-04:
I forgot about `lib_force`. Sure, it should work. If it doesn't work, please open an issue. It is a bug.

https://github.com/platformio/platformio/issues

closed 2016-09-04 by ivankravets