platformio/platformio-libmirror #95: Library installed but empty

zgoda, 2016-10-26:
I'm talking library ID 200 - http://platformio.org/lib/show/200/DigitalIO. Library repository is [hosted on GitHub](https://github.com/greiman/DigitalIO).

I added it to my `platformio.ini`:

```
[env:digispark-tiny]
platform = atmelavr
board = digispark-tiny
framework = arduino
lib_deps =
    DigitalIO
    RF24
```

And I see the installation process succeeds yet the library directory is almost empty:

```
$ LANG=C ls -la .piolibdeps/DigitalIO_ID200
total 32
drwx------ 2 jazg jazg 4096 paź 26 12:29 .
drwxrwxr-x 4 jazg jazg 4096 paź 26 12:29 ..
-rw-rw-r-- 1 jazg jazg  731 wrz  26 20:30 .library.json
-rw-rw-r-- 1 jazg jazg  338 wrz  26 20:30 library.json
```

ivankravets, 2016-10-26:
Thanks, fixed! http://platformio.org/lib/show/200/DigitalIO

closed 2016-10-26 by ivankravets

zgoda, 2016-10-26:
Your response time amazes me. :grinning:

zgoda, 2016-11-23:
Unfortunately this did not fix the issue:

```
In file included from src/main.cpp:4:0:
.piolibdeps/RF24_ID433/RF24.h:23:25: fatal error: DigitalIO.h: No such file or directory
#include <DigitalIO.h>
^
compilation terminated.
*** [.pioenvs/digispark-tiny/src/main.o] Error 1
```

```
$ LANG=C ls -la .piolibdeps/DigitalIO_ID200/
total 32
drwx------ 2 jazg jazg 4096 lis  23 09:22 .
drwxrwxr-x 4 jazg jazg 4096 lis  23 09:22 ..
-rw-rw-r-- 1 jazg jazg  731 wrz  26 20:30 .library.json
-rw-rw-r-- 1 jazg jazg  338 wrz  26 20:30 library.json
```

ivankravets, 2016-11-25:
Thanks, fixed again. Sorry.

zgoda, 2016-11-25:
This time all is OK, thanks! :)