platformio/platformio-libmirror #154: Add MD_Parola

staynless, 2019-06-08:
https://github.com/MajicDesigns/MD_Parola

staynless, 2019-06-08:
tried to add using this code

{
    "name": "MD_PAROLA",
    "keywords": "scrolling,LED, matrix, text, displays",
    "description": "Arduino Library for scrolling LED matrix text displays",
    "repository": 
    {
      "type": "git",
      "url": "https://github.com/MajicDesigns/MD_Parola"
    },
    "authors":
    [
      {
        "name": "Marco Colli",
        "email": "",
        "url": "http://arduinoplusplus.wordpress.com",
        "maintainer": true
      }

    ],
    "dependencies":
    {
      "name": "MD_MAX72xx",
      "authors": "Marco Colli",
      "frameworks": "arduino"
    },
    "version": "2.0",
    "frameworks": "arduino",
    "platforms": "*"
  }
upon command platformio lib register library.json
got error Error: Invalid library config URL 'library.json'

RoliAlpha, 2019-06-10:
You just need to change the url that you registered from "https://github.com/MajicDesigns/MD_Parola/master/library.json" to "https://raw.githubusercontent.com/MajicDesigns/MD_Parola/master/library.json",it will work.I just find it a minute ago.